robosoft-ai/SMACC2 #43: Clean up compilation warnings related to tracing

reelrobotics, 2021-09-22:
Need to get pragma'd

pabloinigoblasco, 2021-09-24:
These are the messages I mean.
You have to build smacc in a clean repo.

[image]

This is the compile line I use, probably it shows more information during the building process:

reset && colcon build --event-handlers console_direct+ --packages-up-to sm_atomic_performance_test

> In file included from /opt/ros/rolling/include/rclcpp/executors/single_threaded_executor.hpp:28,
>                  from /opt/ros/rolling/include/rclcpp/executors.hpp:22,
>                  from /opt/ros/rolling/include/rclcpp/rclcpp.hpp:155,
>                  from /home/geus/Desktop/smacc2/src/SMACC2/smacc2/include/smacc2/common.hpp:40,
>                  from /home/geus/Desktop/smacc2/src/SMACC2/smacc2/include/smacc2/smacc_state.hpp:21,
>                  from /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_state.cpp:15:
> /opt/ros/rolling/include/rclcpp/node.hpp:412:3: note: declared here
>   412 |   declare_parameter(const std::string & name);
>       |   ^~~~~~~~~~~~~~~~~
> In file included from /home/geus/Desktop/smacc2/src/SMACC2/smacc2/include/smacc2/smacc_tracing/smacc_tracing.hpp:15,
>                  from /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:15:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp: In function ‘void ros_trace_smacc_event(const char*)’:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:26:43: warning: unused parameter ‘event_type’ [-Wunused-parameter]
>    26 | void TRACEPOINT(smacc_event, const char * event_type) { CONDITIONAL_TP(smacc_event, event_type); }
>       |                              ~~~~~~~~~~~~~^~~~~~~~~~
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp: In function ‘void ros_trace_update_start(const char*)’:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:28:44: warning: unused parameter ‘updatable_element_name’ [-Wunused-parameter]
>    28 | void TRACEPOINT(update_start, const char * updatable_element_name)
>       |                               ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp: In function ‘void ros_trace_update_end(const char*)’:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:33:42: warning: unused parameter ‘updatable_element_name’ [-Wunused-parameter]
>    33 | void TRACEPOINT(update_end, const char * updatable_element_name)
>       |                             ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp: In function ‘void ros_trace_state_onRuntimeConfigure_start(const char*)’:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:38:62: warning: unused parameter ‘state_name’ [-Wunused-parameter]
>    38 | void TRACEPOINT(state_onRuntimeConfigure_start, const char * state_name)
>       |                                                 ~~~~~~~~~~~~~^~~~~~~~~~
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp: In function ‘void ros_trace_state_onRuntimeConfigure_end(const char*)’:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:43:60: warning: unused parameter ‘state_name’ [-Wunused-parameter]
>    43 | void TRACEPOINT(state_onRuntimeConfigure_end, const char * state_name)
>       |                                               ~~~~~~~~~~~~~^~~~~~~~~~
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp: In function ‘void ros_trace_state_onEntry_start(const char*)’:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:48:51: warning: unused parameter ‘state_name’ [-Wunused-parameter]
>    48 | void TRACEPOINT(state_onEntry_start, const char * state_name)
>       |                                      ~~~~~~~~~~~~~^~~~~~~~~~
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp: In function ‘void ros_trace_state_onEntry_end(const char*)’:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:53:49: warning: unused parameter ‘state_name’ [-Wunused-parameter]
>    53 | void TRACEPOINT(state_onEntry_end, const char * state_name)
>       |                                    ~~~~~~~~~~~~~^~~~~~~~~~
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp: In function ‘void ros_trace_state_onExit_start(const char*)’:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:58:50: warning: unused parameter ‘state_name’ [-Wunused-parameter]
>    58 | void TRACEPOINT(state_onExit_start, const char * state_name)
>       |                                     ~~~~~~~~~~~~~^~~~~~~~~~
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp: In function ‘void ros_trace_state_onExit_end(const char*)’:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:63:48: warning: unused parameter ‘state_name’ [-Wunused-parameter]
>    63 | void TRACEPOINT(state_onExit_end, const char * state_name)
>       |                                   ~~~~~~~~~~~~~^~~~~~~~~~
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp: In function ‘void ros_trace_client_behavior_on_entry_start(const char*, const char*, const char*)’:
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:69:48: warning: unused parameter ‘state_name’ [-Wunused-parameter]
[... cut ...]
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:90:70: warning: unused parameter ‘orthogonal_name’ [-Wunused-parameter]
>    90 |   client_behavior_on_exit_end, const char * state_name, const char * orthogonal_name,
>       |                                                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
> /home/geus/Desktop/smacc2/src/SMACC2/smacc2/src/smacc2/smacc_tracing.cpp:91:16: warning: unused parameter ‘client_behavior_name’ [-Wunused-parameter]
>    91 |   const char * client_behavior_name)

brettpac, 2021-09-29:
I'm thinking this would go well with trace_generator.py
Declan made some progress on this earlier.

pabloinigoblasco, 2021-11-09:
Still around.
these warnings are just part of the c++ building process.
We include some headers of ros2_tracing in one of our header so we inherit these warnings. 
My proposal was just adding for those header files some deactivation of some of these warning errors using #pragma. 

[image]