robosoft-ai/SMACC2 #57: Code optimization discussion

destogl, 2021-09-28:
When analyzing code with Valgrind and looking into it in detail, I found the following (potential) issues. These comments are regarding  `sm_atomic_performance_test` package, but maybe be also present in the other packages.

Code issues:
1. There is no header file guarding in state files — adding `#pragma once` solves this.
2. States are forward-declared in the main `.hpp` file [as classes](https://github.com/robosoft-ai/SMACC2/blob/16a22f62be412ae1ac27784d93a1a8b550ac5f13/smacc2_sm_reference_library/sm_atomic_performance_test/include/sm_atomic_performance_test/sm_atomic_performance_test.hpp#L22-L24), but later as `struct` in their respective files, [State1](https://github.com/robosoft-ai/SMACC2/blob/master/smacc2_sm_reference_library/sm_atomic_performance_test/include/sm_atomic_performance_test/states/st_state_1.hpp), [State2](https://github.com/robosoft-ai/SMACC2/blob/master/smacc2_sm_reference_library/sm_atomic_performance_test/include/sm_atomic_performance_test/states/st_state_2.hpp). This could lead to linking issues on windows.

destogl, 2021-09-28:
Valgrind output from `sm_atomic_performance_test`

```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sm_atomic_performance_test_node-1]: process started with pid [179876]
[sm_atomic_performance_test_node-1] ==179876== Memcheck, a memory error detector
[sm_atomic_performance_test_node-1] ==179876== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
[sm_atomic_performance_test_node-1] ==179876== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
[sm_atomic_performance_test_node-1] ==179876== Command: /home/deniss/workspace/robosoft-ai/smacc_rolling/install/sm_atomic_performance_test/lib/sm_atomic_performance_test/sm_atomic_performance_test_node --ros-args --log-level FATAL --ros-args -r __node:=sm_atomic_performance_test --params-file /home/deniss/workspace/robosoft-ai/smacc_rolling/install/sm_atomic_performance_test/share/sm_atomic_performance_test/config/sm_atomic_config.yaml
[sm_atomic_performance_test_node-1] ==179876== 
[sm_atomic_performance_test_node-1] ==179876== Conditional jump or move depends on uninitialised value(s)
[sm_atomic_performance_test_node-1] ==179876==    at 0x4B1B4D8: smacc2::SignalDetector::pollOnce() (in /home/deniss/workspace/robosoft-ai/smacc_rolling/build/smacc2/libsmacc2.so)
[sm_atomic_performance_test_node-1] ==179876==    by 0x4B1D8BC: smacc2::SignalDetector::pollingLoop() (in /home/deniss/workspace/robosoft-ai/smacc_rolling/build/smacc2/libsmacc2.so)
[sm_atomic_performance_test_node-1] ==179876==    by 0x15F4E6: void smacc2::run<sm_atomic_performance_test::SmAtomicPerformanceTest>() (in /home/deniss/workspace/robosoft-ai/smacc_rolling/build/sm_atomic_performance_test/sm_atomic_performance_test_node)
[sm_atomic_performance_test_node-1] ==179876==    by 0x155AFD: main (in /home/deniss/workspace/robosoft-ai/smacc_rolling/build/sm_atomic_performance_test/sm_atomic_performance_test_node)
[sm_atomic_performance_test_node-1] ==179876== 
[sm_atomic_performance_test_node-1] ==179876== Conditional jump or move depends on uninitialised value(s)
[sm_atomic_performance_test_node-1] ==179876==    at 0x4B1B4F0: smacc2::SignalDetector::pollOnce() (in /home/deniss/workspace/robosoft-ai/smacc_rolling/build/smacc2/libsmacc2.so)
[sm_atomic_performance_test_node-1] ==179876==    by 0x4B1D8BC: smacc2::SignalDetector::pollingLoop() (in /home/deniss/workspace/robosoft-ai/smacc_rolling/build/smacc2/libsmacc2.so)
[sm_atomic_performance_test_node-1] ==179876==    by 0x15F4E6: void smacc2::run<sm_atomic_performance_test::SmAtomicPerformanceTest>() (in /home/deniss/workspace/robosoft-ai/smacc_rolling/build/sm_atomic_performance_test/sm_atomic_performance_test_node)
[sm_atomic_performance_test_node-1] ==179876==    by 0x155AFD: main (in /home/deniss/workspace/robosoft-ai/smacc_rolling/build/sm_atomic_performance_test/sm_atomic_performance_test_node)
[sm_atomic_performance_test_node-1] ==179876==
```

pabloinigoblasco, 2021-09-28:
We may also have a look to the cpu performance using callgrind:
https://www.valgrind.org/docs/manual/cl-manual.html

And also using the graphical tool to analyze its output files: kcachegrind

With that tool we can see what are the functions that are taking longer.
Please, lets do tests in release and also using the log-level=FATAL

pabloinigoblasco, 2021-09-28:
> Valgrind output from `sm_atomic_performance_test`
>
> ```
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [sm_atomic_performance_test_node-1]: process started with pid [179876] [...]

Concerning this, it sounds to me to some variable initialized to the default zero value. Nonetheless, it would be nice to find the way to get which line of code it refers. (I see [here ](https://stackoverflow.com/questions/12175200/conditional-jump-or-move-depends-on-uninitialised-values-on-function-call) that the file and the line of code and the file should be provided )

pabloinigoblasco, 2021-09-28:
> 
> * The struct/class issue it would be nice to solve it, but lets follow the priorities on the pending tasks because does not look critical.

Everything should be at the current moment struct. Probably class is makes more sematic sense because we are inheriting from SmaccState, but the visibility of the methods should be updated if we moved to class.

DecDury, 2021-10-03:
dec@dec-Wild-Dog-Pro:~/workspace/rolling_ws$ ros2 launch sm_atomic_performance_test_a_1 sm_atomic_performance_test_a_1.launch
[INFO] [launch]: All log files can be found below /home/dec/.ros/log/2021-10-02-17-23-31-140880-dec-Wild-Dog-Pro-38894
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sm_atomic_performance_test_a_1_node-1]: process started with pid [38899]
[sm_atomic_performance_test_a_1_node-1] 1633220611.378020 [0] sm_atomic_: using network interface enp0s31f6 (udp/192.168.1.27) selected arbitrarily from: enp0s31f6, docker0
[sm_atomic_performance_test_a_1_node-1] Waited 10001 ms
[sm_atomic_performance_test_a_1_node-1] Number of iterations 1952
[INFO] [sm_atomic_performance_test_a_1_node-1]: process has finished cleanly [pid 38899]

195.2 hz

[image]

DecDury, 2021-10-03:
[sm_atomic_performance_test_a_1_node-1] Waited 10005.4 ms
[sm_atomic_performance_test_a_1_node-1] Number of iterations 1948
[INFO] [sm_atomic_performance_test_a_1_node-1]: process has finished cleanly [pid 40469]

close to 200hz
194.8hz

reelrobotics, 2021-10-06:
https://baptiste-wicht.com/posts/2011/09/profile-c-application-with-callgrind-kcachegrind.html

closed 2021-11-02 by reelrobotics