robosoft-ai/SMACC2 #79: Replace use of unsigned long with something else (uint64?)

reelrobotics, 2021-09-30:
Throughout the library, replace use of unsigned long with something else (uint64?)

reelrobotics, 2021-10-06:
The expectation would be to have an engineer perform a search and replace throughout the library, and then once everything is compiling and running, perhaps performing some minimal runtime testing by running the sm_atomic, sm_atomic_performance_test_a_1, and sm_three_some state machines.
Commands are...

ros2 launch sm_atomic sm_atomic.launch

ros2 launch sm_atomic_performance_test_a_1 sm_atomic_performance_test_a_1.launch

ros2 launch sm_three_some sm_three_some.launch

pabloinigoblasco, 2022-11-21:
This is already done in SMACC2 humble in this PR:
[[](https://github.com/robosoft-ai/SMACC2/pull/335/files)](https://github.com/robosoft-ai/SMACC2/pull/335/files)

By the moment it is not ported to any other SMACC2 or SMACC1's branch

closed 2022-11-24 by sync-by-unito[bot]