robosoft-ai/SMACC2 #176: Adding transition guards?

MrBlenny, 2021-12-09:
Hello hello, thanks again for all your great work!

I've had one feature on my mind and I'm very keen to hear your thoughts. That is **transition guards**. This can partially be achieved with a `SrConditional` state reactor but I think adding guards to the transition table would be a much tidier, more efficient and more flexible solution. I'm unsure of the best way to achieve this...

### Example syntax
```c++
  // TRANSITION TABLE
  typedef boost::mpl::list<
    Transition<EvSomeEvent<CbSomeClientBehavior, OrSomeOrthogonal>, State1, SUCCESS>
    >reactions;
    
  // TRANSITION TABLE WITH CLASS GUARD SYNTAX??
  typedef boost::mpl::list<
    Transition<EvSomeEvent<CbSomeClientBehavior, OrSomeOrthogonal>, GdSomeGuard, State1, SUCCESS>
    >reactions;
```

I'm unsure if passing a `Gd` (guard) class into the template is the best way to achieve this as it *probably?* doesn't need any lifecycle methods. Nonetheless, I can imagine some niche scenarios where a `Gd` class provides a convenient state-container for guard-related logic eg) if the guard prevents a transition > N times, do something such as trigger a Client Behavior.

It may *might* better if it is simple a function:
* Takes in an `Event`. This is particularly useful for things like `EvTopicMessage` which may have `msgData` which is very useful to a guard.
* Some guarding logic - it should have easy access to any state stored in the smacc state, parent state(s) or the configured Orthogonals, Clients or Client Behaviors. This also provides a point to hook into any transition (as there is currently no `onTransition` hook on the smacc state. This is useful for doing things like sending notifications onto a ROS topic once a transition is triggered.

* Returns a bool:
  * `true` - good to transition
  * `false` - don't transition
 
 Thoughts and comments?

pabloinigoblasco, 2021-12-17:
Hello.
We already talked about that idea many months ago. And that is not hard to implement.
Personally I like the proposal and follows the UML state machine standard.

The main argument why we did not implemented that is because it overlapps the funcionality and responsability of the StateReactors that potentially can receive events and generate other events.

Something related is another idea of making any element of SMACC, states, clientbehaviors, clients, and components be able to react to any event directly. That also would be very useful to implement fast reactions based on specific events.

MrBlenny, 2021-12-19:
Very fair. I'm currently working with the `SrConditional` approach. The main downsides are:

* The SMACC RTA display does not work. From memory, it shows something like `internal_event` with no lines from the state reactor inputs. Better RTA support for State Reactors or Guards would help.

* Verbosity. Here is a rather minimal example... I'm not sure if there is a better way to write this. I've had to write others where the state reactors depend on events from other state reactors - at this point, the code gets pretty tricky to follow.

```c++
// TRANSITION TABLE
typedef mpl::list<
    Transition<EvSomeInputEventWithGuard, StGuardedState, GUARDED_STATE>
> reactions;
   
...

void runtimeConfigure()
  {
    auto guard = [ = ](auto * ev) {
        (void)ev;
        
        // as an example, a behavior which contains some state used as a guard
        auto cbSomeBehaviorWithState =
          this->getOrthogonal<OrSomeOrthogonal>()->getClientBehavior<CbSomeBehavior>();

        // guard checks
        if (!cbSomeBehaviorWithState->value) {
          cbPublisher->publish("Cannot transition because the state stored in CbSomeBehavior says so");
          return false;
        }
      
        return true;
      };

    this->createStateReactor<
      SrConditional<EvSomeInputEvent>,
      EvSomeInputEventWithGuard,
      mpl::list<EvSomeInputEvent>
    >(guard);
 }
 ```
* doing things like `cbPublisher->publish` (using the `SmaccPublisherClient`) does not work reliably - I'm not sure why.

closed 2022-02-04 by brettpac