robosoft-ai/SMACC2 #183: [FEA REQUEST] Any support to smacc_service_server_client?

ZhenshengLee, 2021-12-14:
What I found is that smacc provides following client_base:

- action_client
- publisher_client
- service_client
- subscriber_client

Can smacc provide service_server_client?

_Originally posted by @ZhenshengLee in https://github.com/reelrbtx/SMACC/issues/47#issuecomment-993231559_

pabloinigoblasco, 2022-03-17:
Hello @ZhenshengLee 
It is something we have in the scope. It is something we plan to do when we have some gap in the time scheduling.

Nonetheless, any proposal is welcome, anybody could do a PR getting inspiration from the SmaccActionClient or the SubscriberActionClient

yassiezar, 2022-03-29:
@pabloinigoblasco I opened a PR to add this functionality over on the original SMACC1 repo [here](https://github.com/reelrbtx/SMACC/pull/50). Perhaps when you have some time, you can have a look and let me know what you think? I'm not able to port it to SMACC2 just yet, since we're still using ROS Melodic, but I'll be happy to port it once we've moved on to ROS2 and SMACC2

pabloinigoblasco, 2022-03-29:
Hello @yassiezar I had a look to the ROS1 PR thanks for that. I had a look to the code and initially it looks fine to me, I just miss one example to test the funcionality. 

Once it is tested It shouldnt be hard to port it to ROS2.

yassiezar, 2022-03-29:
Ok fair enough. I'll try and add one when I find some time. Would it be fine to adjust `sm_atomic` for it, or is there a better one to put it into?

pabloinigoblasco, 2022-03-29:
Sm atomic is good enough as a base with two states.
One of the states could make a transition when the service server receives a call so we can verify that the service server is working.

ZhenshengLee, 2022-10-28:
close with 193a8a30d431901e8deb7e31f4169a0f3f5d4097

closed 2022-10-28 by ZhenshengLee