robosoft-ai/SMACC2 #259: sm_three_some example not running on galactic

romainreignier, 2022-03-10:
**Describe the bug**
I am currently investigating the SMACC2 usage for a project.
I need subscribers to generate events so I have taken a look at examples using subscribers.
I have tried to run the `sm_three_some` example but there is an error about a topic without name but then, the state machine structure is not printed as for the other examples and after some logs, the node exits with return code 245.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the galactic branch
2. Follow the README to build (vcs import and rosdep install)
3. `colcon build`
4. `source install/setup.bash`
5. `ros2 run sm_three_some sm_three_some_node`

**Expected behavior**
The state machine running

**Output**
<details><summary>console output</summary>

```
$ ros2 run sm_three_some sm_three_some_node 
1646932441.267087 [0] sm_three_s: selected interface "lo" is not multicast-capable: disabling multicast
[INFO] [1646932441.272365375] [SmThreeSome]: Creating State Machine Base: /SmThreeSome
[INFO] [1646932441.272556170] [SmThreeSome]: [sm_three_some::OrTimer] creating client object, type:'cl_ros_timer::ClRosTimer' object tag: 'sm_three_some::OrTimer'
[INFO] [1646932441.272622045] [SmThreeSome]: sm_three_some::OrTimer Orthogonal is created
[INFO] [1646932441.272673740] [SmThreeSome]: [sm_three_some::OrUpdatablePublisher] creating client object, type:'cl_ros_publisher::ClRosPublisher' object tag: 'sm_three_some::OrUpdatablePublisher'
[INFO] [1646932441.272713190] [SmThreeSome]: sm_three_some::OrUpdatablePublisher Orthogonal is created
[INFO] [1646932441.272746011] [SmThreeSome]: [sm_three_some::OrKeyboard] creating client object, type:'cl_keyboard::ClKeyboard' object tag: 'sm_three_some::OrKeyboard'
[INFO] [1646932441.272790118] [SmThreeSome]: [smacc2::ClientHandler<sm_three_some::OrKeyboard, cl_keyboard::ClKeyboard>] Subscribing to topic: /keyboard_unicode
[INFO] [1646932441.273203678] [SmThreeSome]: sm_three_some::OrKeyboard Orthogonal is created
[INFO] [1646932441.273235649] [SmThreeSome]: [sm_three_some::OrSubscriber] creating client object, type:'sm_three_some::cl_subscriber::ClSubscriber' object tag: 'sm_three_some::OrSubscriber'
[ERROR] [1646932441.273276577] [SmThreeSome]: topic client with no topic name set. Skipping subscribing
[INFO] [1646932441.273293049] [SmThreeSome]: sm_three_some::OrSubscriber Orthogonal is created
[INFO] [1646932441.273306483] [SmThreeSome]: [SmaccStateMachine] Introspecting state machine via typeWalker
[INFO] [1646932441.273324766] [SmThreeSome]: Creating State Info: sm_three_some::MsRun
[INFO] [1646932441.273369956] [SmThreeSome]: static OnDefinition: dont exist for sm_three_some::MsRun
[INFO] [1646932441.273392567] [SmThreeSome]: Creating State Info: sm_three_some::StState1
[WARN] [1646932441.273422261] [SmThreeSome]: Real parent state> sm_three_some::MsRun
[INFO] [1646932441.273441264] [SmThreeSome]: EXECUTING ONDEFINITION: sm_three_some::StState1
[INFO] [1646932441.273960627] [static]: [states walking] State StState1client behavior count: 1
[INFO] [1646932441.274002710] [static]: [states walking] State StState1client behavior count: 2
[INFO] [1646932441.274034218] [static]: [states walking] State StState1client behavior count: 3
[INFO] [1646932441.274061452] [static]: [states walking] State StState1client behavior count: 4
[INFO] [1646932441.274070888] [SmThreeSome]: State N13sm_three_some8StState1E Walker has transition list
[INFO] [1646932441.279404674] [SmThreeSome]: State sm_three_some::StState1 Walker transition: cl_ros_timer::EvTimer<cl_ros_timer::CbTimerCountdownOnce, sm_three_some::OrTimer>
[INFO] [1646932441.279432887] [SmThreeSome]: State sm_three_some::StState1 Walker transition: cl_ros_timer::EvTimer<cl_ros_timer::CbTimerCountdownOnce, sm_three_some::OrTimer>
[INFO] [1646932441.279459664] [SmThreeSome]: Transition tag: sm_three_some::StState1::TIMEOUT
[INFO] [1646932441.279482231] [SmThreeSome]: Creating State Info: sm_three_some::StState2
[INFO] [1646932441.279506551] [SmThreeSome]: EXECUTING ONDEFINITION: sm_three_some::StState2
[INFO] [1646932441.279989362] [static]: [states walking] State StState2client behavior count: 1
[INFO] [1646932441.280016353] [static]: [states walking] State StState2client behavior count: 2
[INFO] [1646932441.280039820] [static]: [states walking] State StState2client behavior count: 3
[INFO] [1646932441.280068287] [static]: [states walking] State StState2client behavior count: 4
[INFO] [1646932441.283668636] [SmThreeSome]: State N13sm_three_some8StState2E Walker has transition list
[INFO] [1646932441.288629471] [SmThreeSome]: State sm_three_some::StState2 Walker transition: cl_ros_timer::EvTimer<cl_ros_timer::CbTimerCountdownOnce, sm_three_some::OrTimer>
[INFO] [1646932441.288672206] [SmThreeSome]: State sm_three_some::StState2 Walker transition: cl_ros_timer::EvTimer<cl_ros_timer::CbTimerCountdownOnce, sm_three_some::OrTimer>
[INFO] [1646932441.288709286] [SmThreeSome]: Transition tag: sm_three_some::StState2::TIMEOUT
[INFO] [1646932441.288722180] [SmThreeSome]: Creating State Info: sm_three_some::StState3
[INFO] [1646932441.288816230] [SmThreeSome]: EXECUTING ONDEFINITION: sm_three_some::StState3
[INFO] [1646932441.289371514] [static]: [states walking] State StState3client behavior count: 1
[INFO] [1646932441.289415750] [static]: [states walking] State StState3client behavior count: 2
[INFO] [1646932441.289498967] [static]: [states walking] State StState3client behavior count: 3
[INFO] [1646932441.289543580] [static]: [states walking] State StState3client behavior count: 4
[INFO] [1646932441.289554102] [SmThreeSome]: State N13sm_three_some8StState3E Walker has transition list
[INFO] [1646932441.294809550] [SmThreeSome]: State sm_three_some::StState3 Walker transition: cl_ros_timer::EvTimer<cl_ros_timer::CbTimerCountdownOnce, sm_three_some::OrTimer>
[INFO] [1646932441.294832302] [SmThreeSome]: State sm_three_some::StState3 Walker transition: cl_ros_timer::EvTimer<cl_ros_timer::CbTimerCountdownOnce, sm_three_some::OrTimer>
[INFO] [1646932441.294854379] [SmThreeSome]: Transition tag: sm_three_some::StState3::TIMEOUT
[INFO] [1646932441.294879303] [SmThreeSome]: Creating State Info: sm_three_some::SS1::Ss1
[INFO] [1646932441.294896876] [SmThreeSome]: EXECUTING ONDEFINITION: sm_three_some::SS1::Ss1
[INFO] [1646932441.294919424] [SmThreeSome]: Creating State Info: sm_three_some::SS1::sm_three_some::inner_states::StiState1
[... 109 lines cut ...]
[INFO] [1646932441.472157442] [SmThreeSome]: State sm_three_some::StState1 Walker transition: smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > >
[INFO] [1646932441.472202114] [SmThreeSome]: State sm_three_some::StState1 Walker transition: smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > >
[INFO] [1646932441.472226347] [SmThreeSome]: Transition tag: sm_three_some::StState1::TIMEOUT
$
```

</details>

**Environment:**

*ROS DETAILS*:
- OS: Ubuntu 20.04 LTS
- Version: Galactic

*REPO DETAILS*:
- Repository: robosoft-ai/SMACC2
- Branch: galactic
- Commit: 8ad830

*BUILD DETAILS*
- Build Command: colcon build

*TEST DETAILS*
- Command: ros2 run sm_three_some sm_three_some_node
- Launch file:ros2 launch sm_three_some sm_three_some.launch

romainreignier, 2022-03-10:
Here is the gdb backtrace if of any help
```
Thread 5 "sm_three_some_n" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff5244700 (LWP 2219)]
0x0000555555747be4 in __gnu_cxx::__normal_iterator<std::shared_ptr<smacc2::introspection::TypeInfo>*, std::vector<std::shared_ptr<smacc2::introspection::TypeInfo>, std::allocator<std::shared_ptr<smacc2::introspection::TypeInfo> > > >::__normal_iterator (this=0x7ffff5240ad0, __i=<error reading variable>) at /usr/include/c++/9/bits/stl_iterator.h:807
807           : _M_current(__i) { }
(gdb) bt
#0  0x0000555555747be4 in __gnu_cxx::__normal_iterator<std::shared_ptr<smacc2::introspection::TypeInfo>*, std::vector<std::shared_ptr<smacc2::introspection::TypeInfo>, std::allocator<std::shared_ptr<smacc2::introspection::TypeInfo> > > >::__normal_iterator (this=0x7ffff5240ad0, __i=<error reading variable>) at /usr/include/c++/9/bits/stl_iterator.h:807
#1  0x0000555555742064 in std::vector<std::shared_ptr<smacc2::introspection::TypeInfo>, std::allocator<std::shared_ptr<smacc2::introspection::TypeInfo> > >::begin (this=0x0) at /usr/include/c++/9/bits/stl_vector.h:809
#2  0x000055555573d90f in std::vector<std::shared_ptr<smacc2::introspection::TypeInfo>, std::allocator<std::shared_ptr<smacc2::introspection::TypeInfo> > >::front (this=0x0) at /usr/include/c++/9/bits/stl_vector.h:1121
#3  0x000055555573dd2c in smacc2::introspection::SmaccStateInfo::declareTransition<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > > > (this=0x7ffff0080a70,
    dstState=std::shared_ptr<smacc2::introspection::SmaccStateInfo> (use count 10, weak count 1) = {...}, transitionTag="sm_three_some::StState1::TIMEOUT", transitionType="smacc2::default_transition_tags::ABORT", history=false, transitionTypeInfo=std::shared_ptr<smacc2::introspection::TypeInfo> (empty) = {...})
    at /home/rre/victor_ws/install/smacc2/include/smacc2/introspection/smacc_state_machine_info.hpp:263
#4  0x0000555555738403 in smacc2::introspection::processTransitionAux<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > >, sm_three_some::StState2, sm_three_some::StState1::TIMEOUT> (
    sourceState=std::shared_ptr<smacc2::introspection::SmaccStateInfo> (use count 6, weak count 1) = {...}, history=false, transitionTypeInfo=std::shared_ptr<smacc2::introspection::TypeInfo> (empty) = {...}) at /home/rre/victor_ws/install/smacc2/include/smacc2/introspection/smacc_state_machine_info.hpp:235
#5  0x00005555557335b0 in smacc2::introspection::processTransition<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > >, sm_three_some::StState2, sm_three_some::StState1::TIMEOUT> (t=0x0,
    sourceState=std::shared_ptr<smacc2::introspection::SmaccStateInfo> (use count 6, weak count 1) = {...}) at /home/rre/victor_ws/install/smacc2/include/smacc2/introspection/smacc_state_machine_info.hpp:190
#6  0x000055555572d3af in smacc2::introspection::processTransitions<smacc2::Transition<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > >, sm_three_some::StState2, sm_three_some::StState1::TIMEOUT, boost::statechart::detail::no_context<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > > >, &boost::statechart::detail::no_context<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > > >::no_function> > (sourceState=std::shared_ptr<smacc2::introspection::SmaccStateInfo> (use count 6, weak count 1) = {...}) at /home/rre/victor_ws/install/smacc2/include/smacc2/introspection/smacc_state_machine_info.hpp:344
#7  0x0000555555726da5 in smacc2::introspection::AddTransition::operator()<smacc2::introspection::type_<smacc2::Transition<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > >, sm_three_some::StState2, sm_three_some::StState1::TIMEOUT, boost::statechart::detail::no_context<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > > >, &boost::statechart::detail::no_context<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > > >::no_function> > > (this=0x7ffff5241f80) at /home/rre/victor_ws/install/smacc2/include/smacc2/introspection/smacc_state_machine_info.hpp:416
#8  0x000055555572235b in boost::mpl::aux::for_each_impl<false>::execute<boost::mpl::l_iter<boost::mpl::l_item<mpl_::long_<4l>, smacc2::introspection::type_<smacc2::Transition<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, std_msgs::msg::UInt16_<std::allocator<void> > >, sm_three_some::StState2, sm_three_some::StState1::TIMEOUT, boost::statechart::detail::no_context<smacc2::default_events::EvTopicMessage<sm_three_some::cl_subscriber::ClSubscriber, sm_three_some::OrSubscriber, [... cut ...] t::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> >*>, boost::_bi::value<int> > > >::run (this=0x55555599fbf0) at /usr/include/boost/thread/detail/thread.hpp:120
#32 0x00007ffff7f7443b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#33 0x00007ffff7942609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#34 0x00007ffff75d2293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

romainreignier, 2022-03-10:
For some context, I have started to write a simple state machine based on the sm_atomic_example.
But as soon as I have replaced the timer by a subscriber to produce events, I also have a segfault.

I think that SMACC2 lack simpler examples with just two states but different sources for events, like timer, subscriber, service...

pabloinigoblasco, 2022-03-12:
Hello.
Thanks for your report. We identified the issue you mentioned.
It was a regression in the code, now it should be fixed in all branches.

I hope it helps for you. Please, do not hesitate to re-open the discussion if you are still getting any error.

closed 2022-03-12 by pabloinigoblasco