robosoft-ai/SMACC2 #260: How to trigger an event from a message on a topic?

romainreignier, 2022-03-10:
I would like to trigger an event on message reception on a topic with an empty message type.

I have than created a client:
`cl_empty_subscriber.hpp`
```c++
#pragma once
#include <smacc2/client_bases/smacc_subscriber_client.hpp>
#include <std_msgs/msg/empty.hpp>

namespace sm_sub_example
{
namespace cl_empty_subscriber
{
class ClEmptySubscriber : public smacc2::client_bases::SmaccSubscriberClient<std_msgs::msg::Empty>
{
public:
  ClEmptySubscriber(const std::string& _topic_name)
  : smacc2::client_bases::SmaccSubscriberClient<std_msgs::msg::Empty>(_topic_name)
  {
  }
};
}  // namespace cl_empty_subscriber
}  // namespace sm_sub_example
```

Then a client behavior that do nothing special because as I understand, `SmaccSubscriberClient` already triggers an `EvTopicMessage`:
`cb_empty_subscriber_behavior.hpp`
```c++
#pragma once
#include "sm_sub_example/clients/cl_empty_subscriber/cl_empty_subscriber.hpp"

namespace sm_sub_example
{
namespace cl_empty_subscriber
{
class CbEmptySubscriberBehavior : public smacc2::SmaccClientBehavior
{
public:
  typedef std_msgs::msg::Empty TMessageType;

  void onEntry() { }
};
}  // namespace cl_empty_subscriber
}  // namespace sm_sub_example
```

Now, an orthogonal that set the topic:
`or_empty_subscriber.hpp`
```c++
#pragma once
#include <smacc2/smacc.hpp>
#include "sm_sub_example/clients/cl_empty_subscriber/cl_empty_subscriber.hpp"

namespace sm_sub_example
{
class OrEmptySubscriber : public smacc2::Orthogonal<OrEmptySubscriber>
{
public:
  void onInitialize() override { auto client = this->createClient<cl_empty_subscriber::ClEmptySubscriber>("/next"); }
};
}  // namespace sm_sub_example
```

In the state machine declaration (from `sm_atomic` example), I only add the creation of the orthogonal:
```c++
this->createOrthogonal<OrEmptySubscriber>();
```

And in the `st_state_1.hpp`, I add the transition and the configure_orthogonal statements:
```c++
[... 86 lines cut ...]
    sourceState=std::shared_ptr<smacc2::introspection::SmaccStateInfo> (use count 5, weak count 1) = {...}) at /home/rre/victor_ws/install/smacc2/include/smacc2/introspection/smacc_state_machine_info.hpp:168
#12 0x000055555565a2ec in smacc2::introspection::WalkStatesExecutor<sm_sub_example::State1>::walkStates (parentState=std::shared_ptr<smacc2::introspection::SmaccStateInfo> (use count 5, weak count 1) = {...}, rootInitialNode=true) at /home/rre/victor_ws/install/smacc2/include/smacc2/introspection/smacc_state_machine_info.hpp:461
#13 0x0000555555657c87 in smacc2::introspection::SmaccStateMachineInfo::buildStateMachineInfo<sm_sub_example::State1> (this=0x7ffff007ef40) at /home/rre/victor_ws/install/smacc2/include/smacc2/introspection/smacc_state_machine_info.hpp:470
#14 0x0000555555656c82 in smacc2::ISmaccStateMachine::buildStateMachineInfo<sm_sub_example::State1> (this=0x7ffff0000b60) at /home/rre/victor_ws/install/smacc2/include/smacc2/impl/smacc_state_machine_impl.hpp:601
#15 0x00005555556554b9 in smacc2::SmaccStateMachineBase<sm_sub_example::SmAtomic, sm_sub_example::State1>::initiate_impl (this=0x7ffff0000b60) at /home/rre/victor_ws/install/smacc2/include/smacc2/smacc_state_machine_base.hpp:75
#16 0x000055555563e18b in boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> > >::initiate (this=0x7ffff0000e40) at /usr/include/boost/statechart/event_processor.hpp:44
#17 0x0000555555637d9a in boost::statechart::processor_container<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> >, boost::function0<void>, std::allocator<boost::statechart::none> >::initiate_processor_impl (this=0x7fffffffcfa8, processor=...)
    at /usr/include/boost/statechart/processor_container.hpp:397
#18 0x000055555564e12c in boost::_mfi::mf1<void, boost::statechart::processor_container<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> >, boost::function0<void>, std::allocator<boost::statechart::none> >, boost::weak_ptr<std::auto_ptr<boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> > > > > const&>::operator() (this=0x5555557a39d0, p=0x7fffffffcfa8, a1=...) at /usr/include/boost/bind/mem_fn_template.hpp:165
#19 0x000055555564b79a in boost::_bi::list2<boost::_bi::value<boost::statechart::processor_container<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> >, boost::function0<void>, std::allocator<boost::statechart::none> >*>, boost::_bi::value<boost::weak_ptr<std::auto_ptr<boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> > > > > > >::operator()<boost::_mfi::mf1<void, boost::statechart::processor_container<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> >, boost::function0<void>, std::allocator<boost::statechart::none> >, boost::weak_ptr<std::auto_ptr<boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> > > > > const&>, boost::_bi::list0> (this=0x5555557a39e0, f=..., a=...) at /usr/include/boost/bind/bind.hpp:319
#20 0x0000555555648788 in boost::_bi::bind_t<void, boost::_mfi::mf1<void, boost::statechart::processor_container<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> >, boost::function0<void>, std::allocator<boost::statechart::none> >, boost::weak_ptr<std::auto_ptr<boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> > > > > const&>, boost::_bi::list2<boost::_bi::value<boost::statechart::processor_container<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> >, boost::function0<void>, std::allocator<boost::statechart::none> >*>, boost::_bi::value<boost::weak_ptr<std::auto_ptr<boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> > > > > > > >::operator() (this=0x5555557a39d0) at /usr/include/boost/bind/bind.hpp:1294
#21 0x0000555555643fe1 in boost::detail::function::void_function_obj_invoker0<boost::_bi::bind_t<void, boost::_mfi::mf1<void, boost::statechart::processor_container<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> >, boost::function0<void>, std::allocator<boost::statechart::none> >, boost::weak_ptr<std::auto_ptr<boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> > > > > const&>, boost::_bi::list2<boost::_bi::value<boost::statechart::processor_container<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> >, boost::function0<void>, std::allocator<boost::statechart::none> >*>, boost::_bi::value<boost::weak_ptr<std::auto_ptr<boost::statechart::event_processor<boost::statechart::fifo_scheduler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> > > > > > > >, void>::invoke (function_obj_ptr=...) at /usr/include/boost/function/function_template.hpp:158
#22 0x0000555555621fbb in boost::function0<void>::operator() (this=0x7ffff52594d0) at /usr/include/boost/function/function_template.hpp:763
#23 0x0000555555631211 in boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >::operator() (this=0x7fffffffcfd8, maxItemCount=0) at /usr/include/boost/statechart/fifo_worker.hpp:134
[... cut ...] duler<boost::statechart::fifo_worker<std::allocator<boost::statechart::none> >, std::allocator<boost::statechart::none> >*>, boost::_bi::value<int> > > >::run (this=0x5555557c9b70) at /usr/include/boost/thread/detail/thread.hpp:120
#29 0x00007ffff78cc43b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#30 0x00007ffff794a609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#31 0x00007ffff75e8293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

</details>

pabloinigoblasco, 2022-03-12:
Hello.
Thanks for your report.
The development process you are following in your code looks fine to me.

As in the other issue, there were a regression in the code that should be already fixed.
Please do not hesitate to ask again if something wrong keeps happening.

closed 2022-03-12 by pabloinigoblasco

romainreignier, 2022-03-14:
Thank you for your answer and for fixing the issue.
Nonetheless, in the above example, the event does not seem to be triggered.
Do I need to add some implementation in my subscriber client in order to manually signal the event?
I have thought that `SmaccSubscriberClient` already handles that. Is it the case?