robosoft-ai/SMACC2 #320: [QST] The concept of component.

ZhenshengLee, 2022-10-25:
In NavZ and MoveGroup Client Library, There is more usage of component than Or-Cl-Cb toolchain.

So, what't the arch opinion of component usage? Can component post events?

Thanks.

pabloinigoblasco, 2022-11-03:
Hello @ZhenshengLee 

We have a few different concepts around:
 - orthogonal, clients and components

Clients are easier to understand because typically is the class that controls some remote system (navigation, moveit, etc).
Components are reusable subparts of a client. It is very convenient to use components instead of using a large client object. 

Orthogonals are just containers of clients, they describe a subsystem that runs in parallel. An orthogonal may have one or many clients.

We could be talking for a long time about all these concepts but by the moment you have this:  https://smacc.dev/ . Go to the section of concepts. 

We are also working to improve the documentation.

ZhenshengLee, 2022-11-05:
@pabloinigoblasco  Thanks for your reply!

> We could be talking for a long time about all these concepts but by the moment you have this: https://smacc.dev/ . Go to the section of concepts.

I've already read the all contents of https://smacc.dev/ , and I have more questions about the usage of Or-Cl-Cp-Cb toolchain.

> We are also working to improve the documentation.

this page is lack of the concept of component.

> Clients are easier to understand because typically is the class that controls some remote system (navigation, moveit, etc).

Understood, and Cb is the class to control some remote system too, and the difference between Cb and Cl is Cb can postEv, am I right?

> An orthogonal may have one or many clients.

But in sm_reference_library, there is no demo of multiple clients in one Or, AFAK , am I right?

ZhenshengLee, 2022-11-07:
Thanks to the smacc.dev
Closing this!

closed 2022-11-07 by ZhenshengLee