robosoft-ai/SMACC2 #321: [BUG] std_srvs/srv/empty.hpp: No such file or directory.

ZhenshengLee, 2022-10-28:
**Describe the bug**
cannot find header files.

**To Reproduce**
Steps to reproduce the behavior:
1. colcon build

**Expected behavior**
compile success

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Environment (please complete the following information):**

*ROS DETAILS*:
- OS: Ubuntu 22.04 LTS
- Version: Humble

*REPO DETAILS*:
- Repository: robosoft-ai/SMACC2
- Branch: galactic
- Commit (First 5 Digits):  [193a8a3](https://github.com/robosoft-ai/SMACC2/commit/193a8a30d431901e8deb7e31f4169a0f3f5d4097)

*BUILDSPACE DETAILS (List all packages)*:
- ...

*BUILD DETAILS*
- Build Command: colcon build --symlink-install

**Additional context**
- Anything that may be unusual about your environment

Add any other context about the problem here, especially include any modifications to smacc2 that relate to this issue.

**Error Details**

```sh
smacc2_sm_reference_library/sm_atomic_services/include/sm_atomic_services/clients/cl_service_server.hpp:17:10: fatal error: std_srvs/srv/empty.hpp: No such file or directory
   17 | #include <std_srvs/srv/empty.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

pabloinigoblasco, 2022-11-03:
Thanks @ZhenshengLee 

I think this is already fixed, we are currently working on moving to humble.
Right now we already have an "humble branch", galactic is still the default one though.

ZhenshengLee, 2022-11-04:
I created a fork named SMACC-lite https://github.com/ZhenshengLee/SMACC2-lite with only the core function of SMACC2
The bug exist in this repo and the fix is in https://github.com/ZhenshengLee/SMACC2-lite/commit/87913a30d6703136919e271c7edee890a7b4a4bd

As this repo canbe compiled success, I will close this issue.

Thanks.

closed 2022-11-04 by ZhenshengLee