robosoft-ai/SMACC2 #329: [QST] Must Or-Cb be configured in leaf states?

ZhenshengLee, 2022-11-09:
### description

there is a need when hmi system shutdown the system off, to let state-machine from MsGuardianAutoMode to StManual
So, I created the transition table and configure or-cb toolchain in MsGuardianAutoMode definition hpp
And THAT'S NOT WORKING

```cpp
class MsGuardianAutoMode : public smacc2::SmaccState<MsGuardianAutoMode, SmGuardian, StStandby>
{
public:
   using SmaccState::SmaccState;

   typedef mpl::list<

    Transition<EvDriveModeOff<CbHmi, OrHmi>, StManual, SUCCESS>,
    Transition<EvTakeOver<CbScu, OrScu>, StManual, SUCCESS>

    >reactions;

// STATE FUNCTIONS
    static void staticConfigure()
    {
        // configuring or-cb here is not working
        // must configure or-cb in leaf states
        // configure_orthogonal<OrHmi, CbHmi>();
        // configure_orthogonal<OrScu, CbScu>();

        // debug
        // configure_orthogonal<OrTimer, CbTimerCountdownOnce>(2);
    }
};
```

And when I put the or-cb tool-chain in the leaf state StStandby, The transition defined in MsGuardianAutoMode can be enabled

the StStandby definition is in the below

```cpp
struct StStandby : smacc2::SmaccState<StStandby, MsGuardianAutoMode>
{
    using SmaccState::SmaccState;

// TRANSITION TABLE
    ...
    // debug
    // Transition<EvTimer<CbTimerCountdownOnce, OrTimer>, SS1::SsApa, SUCCESS>
    >reactions;

// STATE FUNCTIONS
    static void staticConfigure()
    {
...

        // top
        configure_orthogonal<OrHmi, CbHmi>();
        configure_orthogonal<OrScu, CbScu>();

        // debug
        // configure_orthogonal<OrTimer, CbTimerCountdownOnce>(5);
    }

};
```

### question

Must Or-Cb be configured in leaf states?

pabloinigoblasco, 2022-11-21:
I think what I think you want to do should work.
However I do not know the internal details of your client behaviors and they could be the reason why that is not working.
Could you please create some executable example we could replicate to identify your issue?

I think what you need to know is the following:
Smacc Events must affects to any level in the current state hierarchy (to MsGuardianAutoMode and also to StStandby ). That means that you could use a client behavior in StStandby that post events that triggers the MsGuardianAutoMode transition table.

ZhenshengLee, 2022-11-22:
> I think what I think you want to do should work.

I agree, but it did not work

> However I do not know the internal details of your client behaviors

The Cb is alwary receive a topic and if the msg.data == 1, it post a event.

> That means that you could use a client behavior in StStandby that post events that triggers the MsGuardianAutoMode transition table.

What I want is to use only a Cb in MsGuardianAutoMode and post events that triggers the transition table in all leaf states of MsGuardianAutoMode including StStandby.

But I found that I have to put Cb, that is `configure_orthogonal<OrHmi, CbHmi>();` in all leaf states of MsGuardianAutoMode, other than MsGuardianAutoMode, and leave the only transition table, that is  `Transition<EvDriveModeOff<CbHmi, OrHmi>, StManual, SUCCESS>,` in the MsGuardianAutoMode.

brettpac, 2023-06-04:
Hello @ZhenshengLee, I'm closing this issue for the time being, but I'm happy to reopen it at some point in the future if we can recreate the issue. Thank you.

closed 2023-06-04 by brettpac