robosoft-ai/SMACC2 #330: [BUG] ClRoslaunch std::async launching thread cannot return and client behavior blocks.

ZhenshengLee, 2022-11-12:
**Describe the bug**

The ClRoslaunch is about to launch a python launch which will call some ros2 service call
Some times the service call in launch file would stuck, so the launch thread would block.

the output shows `waiting for finishing client behavior, before leaving the state. Is the client behavior stuck? requesting force finish`

https://github.com/robosoft-ai/SMACC2/blob/7f889c09993de5941e8189476bcf6bdfb1d98c67/smacc2/src/smacc2/smacc_client_async_behavior.cpp#L78-L82

I've seen the code that launching thread in the client behavior is called by std::async, meaning that it
s a async operation and would return right away.

https://github.com/robosoft-ai/SMACC2/blob/7f889c09993de5941e8189476bcf6bdfb1d98c67/smacc2/src/smacc2/client_bases/smacc_ros_launch_client.cpp#L34-L73

QST, 
1. why this async behavior block? whe this client behavior happen?
1. when this happens, how to recovery the behavior of state machine? Must I kill the node and restart the node?
1. How to avoid this client behavior blocking?

ZhenshengLee, 2022-11-22:
I think it may be related with std::future, because similar behavior/issue happens with https://github.com/robosoft-ai/SMACC2/blob/galactic/smacc2/include/smacc2/client_behaviors/cb_call_service.hpp

https://github.com/robosoft-ai/SMACC2/blob/42177d3c5eef0f869f9648c87eb3af1347291922/smacc2/include/smacc2/client_behaviors/cb_call_service.hpp#L44-L46

https://github.com/robosoft-ai/SMACC2/blob/42177d3c5eef0f869f9648c87eb3af1347291922/smacc2/include/smacc2/client_bases/smacc_service_client.hpp#L58-L64

sometimes the `call()` function cannot return and the Cb could be stuck, thus making state machine stuck.

Crowdedlight, 2022-11-28:
> Good client behavior needs to implement their own mechanisms to be interrupted in the case we are leaving the state.
> We have a solution for this, but it is responsibility of the client behavior implementation to use it correctly:

I am having the same error even with ``isShutdownRequested`` used the same way as you have in the reference example [cb_blinking.hpp](https://github.com/robosoft-ai/SMACC2/blob/d874d1882d3e779a6ccbc0a78bc752fd78d00367/smacc2_sm_reference_library/sm_husky_barrel_search_1/include/sm_husky_barrel_search_1/clients/led_array/client_behaviors/cb_blinking.hpp) 

My Entry looks like this and ``this->checkDistanceToTarget();`` is not blocking. The state is set to continue to the next one as ``checkDistanceToTarget()`` calls ``postSuccessEvent();``.
```c++
  void onEntry() override {

    log_info("OnEntry, Checking pos with 2hz");
    rclcpp::Rate rate(2); // 2Hz

    // remember to check the shutdown request, so we don't block state transitions
    while (!isShutdownRequested()) {
      log_info("Is shutdown requested: " + std::to_string(isShutdownRequested()));
      log_info("Checking distance...");
      this->checkDistanceToTarget();
      rate.sleep();
    }
    log_info("Exited loop...");

  }
```

My log, it will keep repeating these entries which indicates that the function ``this->checkDistanceToTarget();`` keeps being called multiple iterations and the loop never stops. It seems like the ``isShutdownRequested()`` never gets updated. 
```
[navigator_node-2] [WARN] [1669644347.285719186] [navigator]: [navigator::cl_drone_interface::CbTargetPosChecker] waiting for finishing client behavior, before leaving the state. Is the client behavior stuck?
[navigator_node-2] [INFO] [1669645400.507621181] [navigator]:  [CbTargetPosChecker] Is shutdown requested: 0
[navigator_node-2] [INFO] [1669644347.664684620] [navigator]:  [CbTargetPosChecker] Checking distance...
[navigator_node-2] [INFO] [1669644347.664729279] [navigator]:  [CbTargetPosChecker] Dist to target pos: 0.001001, Threshold: 0.1m
[navigator_node-2] [INFO] [1669644347.664740655] [navigator]:  [CbTargetPosChecker] At target, returning success
[navigator_node-2] [INFO] [1669644347.664758714] [navigator]: Event smacc2::EvCbSuccess<navigator::cl_drone_interface::CbTargetPosChecker, navigator::OrDroneInterface>
[navigator_node-2] [WARN] [1669644347.664772301] [navigator]: [ISmaccStateMachine] CURRENT STATE SCOPED EVENT DISCARDED, state is exiting/transitioning smacc2::EvCbSuccess<navigator::cl_drone_interface::CbTargetPosChecker, navigator::OrDroneInterface>
```

Either I am missing something, or it appears that ``isShutdownRequested()`` is not behaving as intended?

pabloinigoblasco, 2022-12-28:
@Crowdedlight your code looks good to me.
When the state is leaving, it must notify all client behaviors to leave via the SmaccAsyncClientBehavior::requestForceFinish() method.

One question, what branch/commit are you using?

pabloinigoblasco, 2022-12-28:
@ZhenshengLee thanks for notifying both problematic use cases (cbLaunch and CbServiceCall)

By the moment I created a new PR for galactic (and also for humble) to improve CbServiceCall and allow interruption of the wating process. 

https://github.com/robosoft-ai/SMACC2/pull/375/files

ZhenshengLee, 2022-12-29:
> https://github.com/robosoft-ai/SMACC2/pull/375/files

once the PR being merged, this canbe closed.

pabloinigoblasco, 2022-12-29:
PR is merged into galactic.
Humble is in progress.

closed 2022-12-29 by ZhenshengLee

pabloinigoblasco, 2022-12-30:
Also merged in humble #379

Crowdedlight, 2023-01-10:
> @Crowdedlight your code looks good to me. When the state is leaving, it must notify all client behaviors to leave via the SmaccAsyncClientBehavior::requestForceFinish() method.
> 
> One question, what branch/commit are you using?

Sorry for the late reply @pabloinigoblasco, vacation and sickness kept me away for a while. 

Is ``SmaccAsyncClientBehavior::requestForceFinish()`` a method I have to explicitly call when I leave the state or is it called automatically by smacc when a state change is triggered? 

We are using branch ``foxy`` atm, but planning to upgrade to ``humble`` within a few months.