robosoft-ai/SMACC2 #369: [BUG] State machine process shutdown with coredump with Ctrl+\(SIGQUIT).

ZhenshengLee, 2022-11-30:
**Describe the bug**
State machine process shutdown with coredump with Ctrl+\(SIGQUIT).

**To Reproduce**
Steps to reproduce the behavior:

simply execute the program

```sh
~/zs_ws/ga_ros/colcon/install/sm_guardian/lib/sm_guardian$ ./sm_guardian_node
```

**Expected behavior**
program canbe shutdown gracefully.

**Screenshots**
[image]

**Environment (please complete the following information):**

*ROS DETAILS*:
- OS: Ubuntu 20.04 LTS
- Version: Humble

*REPO DETAILS*:
- Repository: [e.g. robosoft-ai/SMACC2]

pabloinigoblasco, 2022-12-29:
I have been testing this on galactic with sm_atomic.
I also get that same error.

However, debugging I do not detect any raised exception or crash in our code so I am not sure if this is our fault.
I think that is something expected using SIGQUIT. I tested that with other programs and I get the same "core dumped" error.

for example, try with "sleep 1000"

Nonetheless I am still analyzing this. I opened a PR subscribing to the SIGQUIT signal to see if we can get more insights about the issue.

https://github.com/robosoft-ai/SMACC2/pull/377

ZhenshengLee, 2022-12-30:
From https://ftp.gnu.org/old-gnu/Manuals/glibc-2.2.3/html_node/libc_462.html  I believe there will always a coredump if the SIGQUIT Handler is not installed,

after #377 merged, this canbe closed.

pabloinigoblasco, 2022-12-30:
I am afraid that it keeps happening even if we install SIGQUIT handler :S 

I would not worry too much about this, if that message happen with all linux tools (ls, watch, sleep, etc) , this is not a problem or a bad design.

Nonetheless, I think it is not a bad idea to subscribe to some of this signals in the signalDetector file as we have in #377 so we will merge this anyways.