robosoft-ai/SMACC2 #370: [BUG] state machine process cannot quit with Ctrl-C(SIGINT).

ZhenshengLee, 2022-11-30:
**Describe the bug**
- state machine process doesn't shutdown gracefully with Ctrl-C(SIGINT).

**To Reproduce**
Steps to reproduce the behavior:

simply execute the program

```sh
~/zs_ws/ga_ros/colcon/install/sm_guardian/lib/sm_guardian$ ./sm_guardian_node
```

**Expected behavior**

the state machine can shutdown gracefully 

**Screenshots**

[image]

**Environment (please complete the following information):**

*ROS DETAILS*:
- OS: Ubuntu 20.04 LTS
- Version : Humble

*REPO DETAILS*:
- Repository: [e.g. robosoft-ai/SMACC2]

Add any other context about the problem here, especially include any modifications to smacc2 that relate to this issue.

**Error Details**
```
[INFO] [1669775328.919986906] [SmGuardian]: Event cl_ros_timer::EvTimer<cl_ros_timer::ClRosTimer, sm_guardian::OrTimer>
[INFO] [1669775329.919246807] [SmGuardian]: Event cl_ros_timer::EvTimer<cl_ros_timer::ClRosTimer, sm_guardian::OrTimer>
^C[INFO] [1669775330.071200064] [rclcpp]: signal_handler(signum=2)
^C[INFO] [1669775346.201777741] [rclcpp]: signal_handler(signum=2)
^C[INFO] [1669775346.546807965] [rclcpp]: signal_handler(signum=2)
^C[INFO] [1669775346.830753085] [rclcpp]: signal_handler(signum=2)
^C[INFO] [1669775347.083101996] [rclcpp]: signal_handler(signum=2)
^C[INFO] [1669775347.334747393] [rclcpp]: signal_handler(signum=2)
^C[INFO] [1669775347.578868508] [rclcpp]: signal_handler(signum=2)
```

pabloinigoblasco, 2022-12-29:
We need a reproducible example about this. I do not have access to that sm_guardian example.
I tried with sm_atomic and that does not happen.

ZhenshengLee, 2022-12-30:
the example sm_guardian consists of many topic subscribers , like https://github.com/robosoft-ai/SMACC2/blob/galactic/smacc2/include/smacc2/client_base_components/cp_topic_subscriber.hpp or https://github.com/robosoft-ai/SMACC2/tree/galactic/smacc2_client_library/multirole_sensor_client

and I will try to find sometime to get a simple enough reproducible example about this

pabloinigoblasco, 2023-01-09:
I tested this again, and ctrl + c looks to be working properly for me.

ZhenshengLee, 2023-02-15:
I believe this is about app development rather than the smacc2 framework.
closing this.

ZhenshengLee, 2023-02-15:
Thank you very much! @pabloinigoblasco

closed 2023-02-15 by ZhenshengLee