robosoft-ai/SMACC2 #371: [BUG] ros2launch state machine node cannot shutdown gracefully with Ctrl-C(SIGINT).

ZhenshengLee, 2022-11-30:
**Describe the bug**
this bug is related with https://github.com/robosoft-ai/SMACC2/issues/370
- ros2launch will try to shutdown the process with SIGTERM and SIGKILL and SIGKILL will shutdown the process successfully

**To Reproduce**
Steps to reproduce the behavior:

simply ros2launch the program

```sh
ros2 launch sm_guardian sm_guardian.launch.py
```

**Expected behavior**

the process quit gracefully with Ctrl-C.

**Screenshots**

[image]

**Environment (please complete the following information):**

*ROS DETAILS*:
- OS: Ubuntu 20.04 LTS
- Version: Humble

*REPO DETAILS*:
- Repository: [e.g. robosoft-ai/SMACC2]

**Error Details**
```
zs@zs-vm-2204:~/zs_ws/ga_ros/colcon/install/sm_guardian/lib/sm_guardian$ ros2 launch sm_guardian sm_guardian.launch.py 
[INFO] [launch]: All log files can be found below /home/zs/.ros/log/2022-11-30-10-49-59-907312-zs-vm-2204-332052
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sm_guardian_node-1]: process started with pid [332053]
[sm_guardian_node-1] [WARN] [1669776600.399598560] [SmGuardian]: Signal detector frequency (ros param signal_detector_loop_freq): 20.000000
[sm_guardian_node-1] [WARN] [1669776600.409013337] [SmGuardian]: Real parent state> sm_guardian::MsGuardianAutoMode
[sm_guardian_node-1] [WARN] [1669776600.410788546] [SmGuardian]: Real parent state> sm_guardian::SS1::SsApa
[sm_guardian_node-1] [WARN] [1669776600.417391186] [SmGuardian]: Real parent state> sm_guardian::SS2::SsPilot
[sm_guardian_node-1] [WARN] [1669776600.431863583] [SmGuardian]: State machine base creation:SmGuardian
[sm_guardian_node-1] [WARN] [1669776600.435264300] [StManual]: Entering StManual
[sm_guardian_node-1] [WARN] [1669776600.435316089] [SmGuardian]: [StateMachine] setting state active : sm_guardian::StManual
[sm_guardian_node-1] [WARN] [1669776600.435366705] [StManual]: On Entry!
[sm_guardian_node-1] [WARN] [1669776600.435380072] [SmGuardian]: [CbHmiSystem] onEntry().
[sm_guardian_node-1] [WARN] [1669776600.435403093] [SmGuardian]: [CbScuStatus] onEntry().
[sm_guardian_node-1] [WARN] [1669776600.435416066] [SmGuardian]: [StateMachine] setting state active : sm_guardian::StManual
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ERROR] [sm_guardian_node-1]: process[sm_guardian_node-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [sm_guardian_node-1]: sending signal 'SIGTERM' to process[sm_guardian_node-1]
[ERROR] [sm_guardian_node-1]: process[sm_guardian_node-1] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [sm_guardian_node-1]: sending signal 'SIGKILL' to process[sm_guardian_node-1]
[ERROR] [sm_guardian_node-1]: process has died [pid 332053, exit code -9, cmd '/home/zs/zs_ws/ga_ros/colcon/install/sm_guardian/lib/sm_guardian/sm_guardian_node --ros-args --log-level warn --ros-args'].
```

pabloinigoblasco, 2022-12-28:
Thank you @ZhenshengLee I am having a look to this issue.

pabloinigoblasco, 2023-01-11:
Again, in our case this looks it is working.
After some changes in humble and galactic CbRosLaunch is an interrumpible asynchronous client behavior.
ctrl+c should be working.

ZhenshengLee, 2023-02-15:
I believe this is about app development rather than the smacc2 framework.
closing this.

closed 2023-02-15 by ZhenshengLee