robosoft-ai/SMACC2 #508: Build Errors

brettpac, 2023-06-06:
There are some warnings coming from the ROS Buildfarm...

SMACC Msgs
https://build.ros2.org/job/Hbin_ujv8_uJv8__smacc2_msgs__ubuntu_jammy_arm64__binary/35/display/redirect
https://build.ros2.org/job/Hbin_uJ64__smacc2_msgs__ubuntu_jammy_amd64__binary/36/display/redirect

SMACC
https://build.ros2.org/job/Hdev__smacc2__ubuntu_jammy_amd64/23/display/redirect?page=changes
https://build.ros2.org/job/Hdev__smacc2__ubuntu_jammy_amd64/22/display/redirect?page=changes

brettpac, 2023-06-06:
My take on the problem;)...
https://answers.ros.org/question/368106/c-fatal-moveit-2-error-killed-signal-terminated-program-cc1plus-compilation/

pabloinigoblasco, 2023-06-14:
> My take on the problem;)... https://answers.ros.org/question/368106/c-fatal-moveit-2-error-killed-signal-terminated-program-cc1plus-compilation/

Yes, I provided that answer to that rosanswer some time ago. It's a common problem I ve been facing for years, especially in large projects. Unfortunately, controlling the compilation process is not currently available for rosbuild farm (as far as I know).

One potential solution is to break the project down into smaller components, like core and examples. This approach could help alleviate the issue.

As for the specific error with smacc_msgs, it's a bit strange that it doesn't occur when using GitHub actions on a clean machine. To address this, it might be worth considering improvements to the CMakefiles to resolve the problem.

nuclearsandwich, 2023-06-16:
> As for the specific error with smacc_msgs, it's a bit strange that it doesn't occur when using GitHub actions on a clean machine. To address this, it might be worth considering improvements to the CMakefiles to resolve the problem.

There appears to be an un-expressed test dependency for the humble branch of `smacc2_msgs`. This can often be masked during CI if the entire repository's dependencies are resolved together and another package in the repository depends on the package. I've proposed https://github.com/robosoft-ai/SMACC2/pull/516
 to resolve the missing dependency declaration.

nuclearsandwich, 2023-06-16:
I'm not sure whether you would prefer a separate issue for the report below or not. Please let me know if so and I'll open the issue.

On Rolling the SMACC2 package is failing to build but it does not appear to be a resource usage issue like the Humble failures linked above. Here is a snippet from the log output of https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__smacc2__ubuntu_jammy_amd64__binary/92/

```
00:04:30.200 [  7%] Building CXX object CMakeFiles/smacc2.dir/src/smacc2/client.cpp.o
00:04:30.257 /usr/lib/ccache/c++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_THREAD_DYN_LINK -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DFASTCDR_DYN_LINK -DRCUTILS_ENABLE_FAULT_INJECTION -Dsmacc2_EXPORTS -I/tmp/binarydeb/ros-rolling-smacc2-0.4.0/include -isystem /opt/ros/rolling/include/rclcpp -isystem /opt/ros/rolling/include/smacc2_msgs -isystem /opt/ros/rolling/include/rclcpp_action -isystem /opt/ros/rolling/include/tracetools -isystem /opt/ros/rolling/include/ament_index_cpp -isystem /opt/ros/rolling/include/libstatistics_collector -isystem /opt/ros/rolling/include/builtin_interfaces -isystem /opt/ros/rolling/include/rosidl_runtime_c -isystem /opt/ros/rolling/include/rcutils -isystem /opt/ros/rolling/include/rosidl_typesupport_interface -isystem /opt/ros/rolling/include -isystem /opt/ros/rolling/include/rosidl_runtime_cpp -isystem /opt/ros/rolling/include/rosidl_typesupport_fastrtps_cpp -isystem /opt/ros/rolling/include/rmw -isystem /opt/ros/rolling/include/rosidl_dynamic_typesupport -isystem /opt/ros/rolling/include/rosidl_typesupport_fastrtps_c -isystem /opt/ros/rolling/include/rosidl_typesupport_introspection_c -isystem /opt/ros/rolling/include/rosidl_typesupport_introspection_cpp -isystem /opt/ros/rolling/include/rcl -isystem /opt/ros/rolling/include/rcl_interfaces -isystem /opt/ros/rolling/include/service_msgs -isystem /opt/ros/rolling/include/rcl_logging_interface -isystem /opt/ros/rolling/include/rcl_yaml_param_parser -isystem /opt/ros/rolling/include/type_description_interfaces -isystem /opt/ros/rolling/include/rcpputils -isystem /opt/ros/rolling/include/statistics_msgs -isystem /opt/ros/rolling/include/rosgraph_msgs -isystem /opt/ros/rolling/include/rosidl_typesupport_cpp -isystem /opt/ros/rolling/include/rosidl_typesupport_c -isystem /opt/ros/rolling/include/action_msgs -isystem /opt/ros/rolling/include/unique_identifier_msgs -isystem /opt/ros/rolling/include/std_msgs -isystem /opt/ros/rolling/include/rcl_action -g -O2 -ffile-prefix-map=/tmp/binarydeb/ros-rolling-smacc2-0.4.0=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Wall -Wextra -Wpedantic -MD -MT CMakeFiles/smacc2.dir/src/smacc2/client.cpp.o -MF CMakeFiles/smacc2.dir/src/smacc2/client.cpp.o.d -o CMakeFiles/smacc2.dir/src/smacc2/client.cpp.o -c /tmp/binarydeb/ros-rolling-smacc2-0.4.0/src/smacc2/client.cpp
00:04:30.258 In file included from /usr/include/boost/bind.hpp:30,
00:04:43.166                  from /usr/include/boost/statechart/fifo_worker.hpp:14,
00:04:43.166                  from /usr/include/boost/statechart/fifo_scheduler.hpp:12,
00:04:43.166                  from /usr/include/boost/statechart/asynchronous_state_machine.hpp:12,
00:04:43.166                  from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/common.hpp:23,
00:04:43.166                  from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/introspection/smacc_state_machine_info.hpp:28,
00:04:43.166                  from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/introspection/introspection.hpp:297,
00:04:43.166                  from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/impl/smacc_state_machine_impl.hpp:27,
00:04:43.166                  from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/impl/smacc_client_impl.hpp:23,
00:04:43.166                  from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/src/smacc2/client.cpp:21:
00:04:43.166 /usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
00:04:43.166    36 | BOOST_PRAGMA_MESSAGE(
00:04:43.166       | ^~~~~~~~~~~~~~~~~~~~
00:04:43.166 In file included from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/impl/smacc_state_machine_impl.hpp:47,
00:04:43.166                  from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/impl/smacc_client_impl.hpp:23,
00:04:43.166                  from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/src/smacc2/client.cpp:21:
00:04:43.166 /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/smacc_tracing/smacc_tracing.hpp:27:21: error: expected constructor, destructor, or type conversion before ‘(’ token
00:04:43.166    27 |   DECLARE_TRACEPOINT(spinOnce)
00:04:43.166       |                     ^
00:04:43.166 In file included from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/impl/smacc_client_impl.hpp:23,
00:04:43.166                  from /tmp/binarydeb/ros-rolling-smacc2-0.4.0/src/smacc2/client.cpp:21:
00:04:43.166 /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/impl/smacc_state_machine_impl.hpp: In member function ‘void smacc2::ISmaccStateMachine::createOrthogonal()’:
[... cut ...]
00:04:43.166 /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/impl/smacc_state_machine_impl.hpp: In member function ‘void smacc2::ISmaccStateMachine::postEvent(EventType*, smacc2::EventLifeTime)’:
00:04:43.166 /tmp/binarydeb/ros-rolling-smacc2-0.4.0/include/smacc2/impl/smacc_state_machine_impl.hpp:180:3: error: ‘ros_trace_smacc2_event’ was not declared in this scope
00:04:43.166   180 |   TRACEPOINT(smacc2_event, eventtypename);
00:04:43.166       |   ^~~~~~~~~~
00:04:43.166 make[4]: *** [CMakeFiles/smacc2.dir/build.make:93: CMakeFiles/smacc2.dir/src/smacc2/client.cpp.o] Error 1
```

brettpac, 2023-06-17:
Hi @nuclearsandwich,
We've still got some problems. Any suggestions?

https://build.ros2.org/job/Hbin_ujv8_uJv8__smacc2_msgs__ubuntu_jammy_arm64__binary/
https://build.ros2.org/job/Hbin_uJ64__smacc2_msgs__ubuntu_jammy_amd64__binary/

brettpac, 2023-07-30:
Finally working

closed 2023-07-30 by brettpac