robosoft-ai/SMACC2 #513: How to use Action events

ahmadabouelainein, 2023-06-12:
Hi,
I am trying to utilize EvActionSucceeded and EvActionAborted with the Fibonacci action but I couldn't find any documentation or examples using on how I could use the events. Could it be briefly explained what the client and client behavior files should look like?

brettpac, 2023-06-12:
Hi Ahmad, can you give some context with regards to your application? What other packages you're interacting with if any (Nav2 MoveIt etc), style of robot (arms, no arms), what you want the robot to do (generally)? 

I get that you're using this message... https://docs.ros2.org/foxy/api/test_msgs/action/Fibonacci.html
Thank you,

ahmadabouelainein, 2023-06-12:
I'm building on the sm_three_some reference library package. I'm trying to add a client and client behavior to call the fibonacci ros2 action. They seem to work as the action is called and the response is received, but I need to activate a transition to the next state based on the success/failure of the ros2 action. I'm using the predefined EvActionSucceeded event but the transition is not triggered even though the action is succesful. I'm not using any other packages currently as I'm still just exploring the package.

brettpac, 2023-06-14:
Hi Ahmad,
We typically usually use the event EvCbSuccess for our examples. Both EvCbSuccess and EvActionSucceeded happen when we receive the result of the action client, and then we post one or the other. 

Which gives us a hypothesis of where your problem might be occuring... smacc_action_client.hpp line 293

Would it be possible to share your application code with us? Possibly through a fork? or privately?
That way we could see what you're doing.

But if not, a first step would be to try using EvCbSuccess for the client behavior.

Thank you,

ahmadabouelainein, 2023-06-14:
Hi Brett,
Thank you for your response. I thought that the callback function was not being triggered so your hypothesis is similar to mine. I will try the EvCbSuccess and get back to you.

ahmadabouelainein, 2023-06-15:
Hi,
Turns out I mistakenly used the class `SmaccClientBehavior` instead of `SmaccAsyncClientBehavior`. The class change made it it work with EvCbSuccess.

closed 2023-06-15 by ahmadabouelainein