robosoft-ai/SMACC2 #521: warning format occorred.

hirotaka001, 2023-07-25:
**Describe the bug**
When I run colcon build I get the following error.
I think `%lu` is better than `%d`.

**Error Details**
```
ws_humble/src/SMACC2/smacc2_client_library/nav2z_client/nav2z_client/src/nav2z_client/components/waypoints_navigator/cp_waypoints_navigator.cpp:416:37: warning: format ‘%d’ expects argument of type ‘int’, but argument 5 has type ‘uint64_t’ {aka ‘long unsigned int’} [-Wformat=]
  416 |           RCLCPP_ERROR(getLogger(), "parsing waypoint file, syntax error in point %d", i);
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~
      |                                                                                        |
      |                                                                                        uint64_t {aka long unsigned int}
ws_humble/src/SMACC2/smacc2_client_library/nav2z_client/nav2z_client/src/nav2z_client/components/waypoints_navigator/cp_waypoints_navigator.cpp:416:84: note: format string is defined here
      |           RCLCPP_ERROR(getLogger(), "parsing waypoint file, syntax error in point %d", i);
      |                                                                                   ~^
      |                                                                                    |
      |                                                                                    int
      |                                                                                   %ld

ws_humble/src/SMACC2/smacc2_client_library/nav2z_client/nav2z_client/src/nav2z_client/components/waypoints_navigator/cp_waypoints_navigator.cpp:479:37: warning: format ‘%d’ expects argument of type ‘int’, but argument 5 has type ‘uint64_t’ {aka ‘long unsigned int’} [-Wformat=]
  479 |           RCLCPP_ERROR(getLogger(), "parsing waypoint file, syntax error in point %d", i);
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~
      |                                                                                        |
      |                                                                                        uint64_t {aka long unsigned int}
ws_humble/src/SMACC2/smacc2_client_library/nav2z_client/nav2z_client/src/nav2z_client/components/waypoints_navigator/cp_waypoints_navigator.cpp:479:84: note: format string is defined here
  479 |           RCLCPP_ERROR(getLogger(), "parsing waypoint file, syntax error in point %d", i);
      |                                                                                   ~^
      |                                                                                    |
      |                                                                                    int
      |                                                                                   %ld
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'ws_humble'
2. Do 'colcon build'
3. See error

**Environment (please complete the following information):**

*ROS DETAILS*:
- OS: Ubuntu 22.04 LTS
- Version: humble

*REPO DETAILS*:
- Repository: robosoft-ai/SMACC2
- Branch: humble

*BUILD DETAILS*
- Build Command: colcon build

brettpac, 2023-07-30:
Hi, @hirotaka001. Thanks for the issue. Seems related to this...https://github.com/ros2/rclcpp/issues/1743 and https://github.com/robosoft-ai/SMACC2/issues/79

I'm thinking we should change all instances of uint64_t in the repo with int64_t. Would you like to do it and create a pull request?

brettpac, 2023-08-03:
Solved in https://github.com/robosoft-ai/SMACC2/pull/526

closed 2023-08-03 by brettpac