robosoft-ai/SMACC2 #535: Adding manually compiled SMACC2 packages files to /opt/foxy folder

cagdasyilmaz, 2024-01-18:
I am interested in utilizing the "SMACC2". While I am aware that official support for the Foxy distribution has concluded, I am currently working on Ubuntu 20.04 LTS.

To address this, I have successfully built the "foxy" branch of the SMACC2 library from the GitHub repository by following the provided instructions https://github.com/robosoft-ai/SMACC2.

My workspace, located at `~/workspace/foxy_ws`, consists of the `build`, `install`, and `log` folders, in addition to the src folder containing the downloaded source files from the aforementioned GitHub page. Now, my objective is to copy or move each package from the install directory to the `/opt/foxy` directory, ensuring that all the subfolders (i.e., `include`, `lib`, `share`) are correctly distributed within the `/opt/foxy` directory.

Here is the current structure of my workspace:

```
user@user:~/workspace/foxy_ws$ ls
build  install  log  src
```

and install includes:

```
user@user:~/workspace/foxy_ws/install$ ls -l

... COLCON_IGNORE
... local_setup.bash
... local_setup.ps1
... local_setup.sh
... _local_setup_util_ps1.py
... _local_setup_util_sh.py
... local_setup.zsh
... ros_timer_client
... setup.bash
... setup.ps1
... setup.sh
... setup.zsh
... smacc2
... smacc2_msgs
... sm_atomic
... sm_atomic_mode_states
```
I am uncertain about how to incorporate packages such as `ros_timer_client`, `smacc2`, `smacc2_msgs`, `sm_atomic`, and `sm_atomic_mode_states` into any workspace without being confined to `~/workspace/foxy_ws` in my implementations.

If anyone has encountered a similar question or has insights into this matter, I would greatly appreciate your guidance. I have conducted a search on internet but have not found a satisfactory answer.

Thank you in advance.

I have tried to run the following command in the `~/workspace/foxy_ws`

`sudo colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --install-base /opt/foxy`

and it destroyed the destroy the integrity of ros2 framework. I could not run the fundamental `ros2 run` commands after that.

pabloinigoblasco, 2024-01-19:
Generally, the /opt/ros/foxy directory is reserved for packages installed from Debian repositories. Directly copying or modifying files in /opt/ros/foxy is not standard practice and can lead to complications.

For your specific situation, consider these two alternatives:

 * **Overlapping Workspaces**: You can source multiple workspaces sequentially in your bash terminal. This approach is useful if you're working with distinct sets of packages, like SMACC and Foxy. By sourcing the setup.bash of each workspace, you effectively overlap them, which allows you to access packages from both. This way, you can load as many workspaces as needed without altering the system's ROS installation. The command would be:

```bash
source /path/to/workspace1/install/setup.bash
source /path/to/workspace2/install/setup.bash
```

 * **Building Foxy from Source**: Another approach is to build ROS Foxy from source and include the SMACC2 repository within your ROS base workspace. This method creates a complete Foxy workspace containing all the necessary packages, including SMACC2. This approach is more holistic and maintains a clear separation from the system-installed ROS.

However, if you have a specific reason to directly extend or override the contents of /opt/ros/foxy (which is unusual or simply incorrect), you can do so by using Colcon's --install-base option. The command would be:

bash
```bash
colcon build --install-base /opt/ros/foxy
```
Be aware that this might require changing the write permissions of the /opt/ros/foxy folder, which involves administrative privileges and should be done cautiously.

cagdasyilmaz, 2024-01-22:
Dear Pablo,

Thank you for your prompt and explanatory response. You have written a well-documented implementation that is  a means to an end.

closed 2024-01-22 by cagdasyilmaz