robosoft-ai/SMACC2 #537: Error!! when build package

PoommipatWat, 2024-02-29:
[image]

I cannot build package. What should i do?

- ubuntu 22.04LTS
- ros2 humble
- and ryzen 5 + rtx 3060

brettpac, 2024-02-29:
Hmmm. sm_multi_stage_1 is a very large state machine, makes me think that there is sometime of timeout set on the compiler.

Options...
- Run colcon again - sometimes this works.
- Fork and remove sm_multi_stage_1 from the reference library
- Check the log files and post here. Should be somewhere like: log/latest/sm_multi_stage_1/stout.log

pabloinigoblasco, 2024-02-29:
That is a memory limit issue.
Use:

colcon build --parallel-workers 1

Also close other unnecessary processes

brettpac, 2024-03-03:
@PoommipatWat did the colcon build --parallel-workers 1 on a quiet system end up working?