robosoft-ai/SMACC2 #539: error during installation SMACC2 for rolling

dkorenkovnl, 2024-04-25:
**Describe the bug**
I am trying to install [SMACC2](https://github.com/robosoft-ai/SMACC2) for rolling release
During execution the following command
`vcs import src --skip-existing --input src/SMACC2/SMACC2-not-released.rolling.repos`

I am getting the following output
`
E...E....E....
=== src/Universal_Robots_Client_Library (git) ===
Cloning into '.'...
=== src/Universal_Robots_Description (git) ===
Cloning into '.'...
=== src/Universal_Robots_ROS2_Driver (git) ===
Cloning into '.'...
=== src/Universal_Robots_ROS2_Gazebo_Simulation (git) ===
Cloning into '.'...
=== src/Universal_Robots_ROS2_Ignition_Simulation (git) ===
Cloning into '.'...
=== src/autoware_auto_msgs (git) ===
Cloning into '.'...
=== src/gazebo_ros2_control (git) ===
Cloning into '.'...
=== src/husky (git) ===
Could not checkout ref 'fixes-for-smacc2': fatal: invalid reference: fixes-for-smacc2
=== src/ign_ros2_control (git) ===
Could not checkout ref 'main': fatal: invalid reference: main
=== src/navigation2 (git) ===
Cloning into '.'...
=== src/ros2_control_demos (git) ===
Cloning into '.'...
=== src/ros_ign (git) ===
Cloning into '.'...
=== src/slam_toolbox (git) ===
Cloning into '.'...
=== src/ur_msgs (git) ===
Could not checkout ref 'ros2': fatal: invalid reference: ros2

'

*ROS DETAILS*:
- OS: Ubuntu 22.04.4 LTS
- Version :Rolling

*REPO DETAILS*:
- Repository: [e.g. robosoft-ai/SMACC2]
- Branch: rolling
- Commit (First 5 Digits): 4f5c01c111d317799399a37aa242e99e8c9c0a74

**Error Details**
there is no such branch fixes-for-smacc2 in the repo https://github.com/StoglRobotics-forks/husky.git
other errors are simular
```
    type: git
    url: https://github.com/StoglRobotics-forks/husky.git
    version: fixes-for-smacc2
```

brettpac, 2024-05-10:
Hi @dkorenkovnl,
We do accept pull requests:)

Rolling definitely needs a little clean-up. But I think it would be easy.
At the moment, we aren't supporting it though. At least not for a little while.

Too many changes constantly creating little tickets, and for what? - Just my opinion.
You up for a pull request?

Thank you

brettpac, 2024-06-12:
Hi @dkorenkovnl,
Ok, so if you're up for a PR in the future, please lemme know and I'm happy to help.
But at the moment, I'm going close this one out with the line that we're not supporting Rolling at the moment because the bang for the support buck is just not there imho.

Cheers,

closed 2024-06-12 by brettpac