robosoft-ai/SMACC2 #550: Unit and Functional Testing in SMACC2

NithishkumarS, 2024-07-18:
I have been exploring the SMACC2 framework and am looking for some guidance on the best practices for testing, specifically regarding unit testing and functional testing. I have a few questions:

Unit Testing:

1.How should unit tests be structured in SMACC2, especially when there are dependencies on orthogonals, clients, or other components?
2. Are there any recommended approaches or tools for mocking these dependencies to isolate the unit under test?

Integration Testing:

1. What is the recommended way to conduct functional tests that involve multiple orthogonals, clients, and their interactions?
2. How can transitions between states be tested effectively in a functional test? Are there specific frameworks or tools that are particularly useful for this purpose in the context of SMACC2?

Any examples, documentation, or tips you can provide would be greatly appreciated. Thank you for your support.

**Environment (please complete the following information):**

*ROS DETAILS*
- OS:  Ubuntu 22.04 LTS
- Version Humble

NithishkumarS, 2024-10-02:
Are there any recommendations for testing in SMACC2?

RobotBramhana, 2025-02-05:
@NithishkumarS I don't think that's possible, Ideally, we have to create a library for unit tests in SMACC function from which which can then the called by the applications developed on top of it. From the code that I went through, I think it is not possible as of now. One possibility is that, the open source community will have to add those changes. Else, you will have to depend on the dev tools and manual testing. That's how we did it when we worked with SMACC1 4 years ago.

brettpac, 2026-06-08:
Testing in SMACC2 is a topic that comes up regularly, and it deserves a thorough answer rather than a quick one. 
    
SMACC2's primary testing mechanism is integration-level: you launch a state machine, observe its behavior through the ROS2 topic interface, and verify that it behaves correctly. The framework exposes several topics specifically designed for this purpose:

  - /[sm_name]/smacc/status — the current active state, including any superstates or mode states
  - /[sm_name]/smacc/transition_log — a chronological record of every state transition
  - /[sm_name]/smacc/event_log — every event that fired, including which orthogonal posted it
  - /[sm_name]/smacc/state_machine_description — the full static topology of the state machine

The reference library contains several state machines built specifically for this style of testing: sm_cl_ros2_timer_unit_test_1, sm_cl_keyboard_unit_test_1, and sm_coretest_transition_speed_1. Each exercises a specific client or framework behavior and can be verified programmatically by subscribing to the transition log and asserting that expected state names appear in the correct order within a timeout.

For external stimulus injection during tests, the keyboard unicode topic (/keyboard_unicode, std_msgs/msg/UInt16) works well — it lets test scripts drive state transitions without a human at a keyboard. Any other client input can similarly be simulated by publishing to the appropriate topic.

closed 2026-06-08 by brettpac