robosoft-ai/SMACC2 #559: Header dependency missing

emanodus, 2024-09-19:
The header file here https://github.com/robosoft-ai/SMACC2/blob/humble/smacc2/include/smacc2/impl/smacc_orthogonal_impl.hpp
is unable to resolve the symbol demangeledtypes which is defined in samcc2::intropection. 

I would be best if all header files are self contained to avoid confusion in which headers to include.

brettpac, 2024-09-19:
Hi @emanodus can you be provide us with some more specifics please? Commands used, terminal output, system specifics, etc.

emanodus, 2024-09-20:
@brettpac 
The things you requested for, for the most part, would be irrelevant to this issue. Second, I dont have the error anymore as i reolved it by including dependecy headers appropriately on my side. But i thought it would be meaningful to share this here.

 I can describe in breif for more context to the issue. So I am extending SMACC2 for my specific use case, where I had to derive a new class from smacc2::Orthogonal<> class. For this I had to include this header "smacc_orthogonal_imp.hpp" in the header of my derived class. While I was building my derived class as a library I encountered an unresolved symbol error for "demangledTypeName" from here [link](https://github.com/robosoft-ai/SMACC2/blob/1fabc680d0f752b599029099592f5762aab49b61/smacc2/include/smacc2/impl/smacc_orthogonal_impl.hpp#L37C29-L37C46) . To resolve it I simply included impl/introspection/introspection.hpp in my derived class header. 
 
I believe it would have been great if the smacc_orthogonal_impl.hpp header had included all headers which it depends on. Shouldn't all headers be self contained with all its dependecies? Please share your views. (I am afraid I am missing some rationale here)

brettpac, 2024-11-01:
Hi @emanodus do you have a pull request we can look at?
We've been looking into this, but it would be much easier if we could just see the code. Thank you,

closed 2025-11-02 by brettpac