robosoft-ai/SMACC2 #577: This repository has ~250k lines of CHANGELOG

tfoote, 2025-06-10:
This has bogged down the rosindex and will cause many other tools to fail. https://github.com/ros-infrastructure/rosindex/issues/543

A brief overview of them looks like below: 

Most of the content doesn't look like useful content for someone reviewing the changelogs and this is a bunch of content requred for every checkout. 

```
SMACC2$ find . -name CHANGELOG* -exec wc -l {} \;
12967 ./smacc2_performance_tools/performance_tests/sm_atomic_subscribers_performance_test/CHANGELOG.rst
3821 ./smacc2_performance_tools/performance_tests/sm_atomic_performance_trace_1/CHANGELOG.rst
3821 ./smacc2_performance_tools/performance_tests/sm_coretest_transition_speed_1/CHANGELOG.rst
30 ./smacc2/CHANGELOG.rst
28 ./smacc2_msgs/CHANGELOG.rst
81 ./smacc2_event_generator_library/eg_conditional_generator/CHANGELOG.rst
81 ./smacc2_event_generator_library/eg_random_generator/CHANGELOG.rst
81 ./smacc2_state_reactor_library/sr_event_countdown/CHANGELOG.rst
3886 ./smacc2_state_reactor_library/sr_conditional/CHANGELOG.rst
3880 ./smacc2_state_reactor_library/sr_all_events_go/CHANGELOG.rst
12969 ./smacc2_sm_reference_library/sm_pubsub_1/CHANGELOG.rst
3880 ./smacc2_sm_reference_library/sm_respira_1/CHANGELOG.rst
9 ./smacc2_sm_reference_library/sm_dance_bot_strikes_back/CHANGELOG.rst
13003 ./smacc2_sm_reference_library/sm_dance_bot_warehouse_2/CHANGELOG.rst
81 ./smacc2_sm_reference_library/sm_atomic_24hr/CHANGELOG.rst
13064 ./smacc2_sm_reference_library/sm_dance_bot/CHANGELOG.rst
9397 ./smacc2_sm_reference_library/sm_multi_ur5_sim/CHANGELOG.rst
12983 ./smacc2_sm_reference_library/sm_dance_bot_warehouse/CHANGELOG.rst
3880 ./smacc2_sm_reference_library/sm_branching/CHANGELOG.rst
22363 ./smacc2_sm_reference_library/sm_test_moveit_ur5_sim/CHANGELOG.rst
3881 ./smacc2_sm_reference_library/sm_three_some/CHANGELOG.rst
12967 ./smacc2_sm_reference_library/sm_advanced_recovery_1/CHANGELOG.rst
55 ./smacc2_sm_reference_library/sm_atomic_services/CHANGELOG.rst
12967 ./smacc2_sm_reference_library/sm_multi_stage_1/CHANGELOG.rst
87 ./smacc2_sm_reference_library/sm_pack_ml/CHANGELOG.rst
9177 ./smacc2_sm_reference_library/sm_aws_warehouse_navigation/CHANGELOG.rst
3823 ./smacc2_sm_reference_library/sm_autoware_avp/CHANGELOG.rst
3880 ./smacc2_sm_reference_library/sm_atomic/CHANGELOG.rst
13027 ./smacc2_sm_reference_library/sm_dance_bot_warehouse_3/CHANGELOG.rst
113 ./smacc2_sm_reference_library/sm_atomic_hierarchy/CHANGELOG.rst
3921 ./smacc2_client_library/keyboard_client/CHANGELOG.rst
13084 ./smacc2_client_library/nav2z_client/nav2z_client/CHANGELOG.rst
3830 ./smacc2_client_library/nav2z_client/custom_planners/nav2z_planners_common/CHANGELOG.rst
3830 ./smacc2_client_library/nav2z_client/custom_planners/undo_path_global_planner/CHANGELOG.rst
3830 ./smacc2_client_library/nav2z_client/custom_planners/backward_local_planner/CHANGELOG.rst
3828 ./smacc2_client_library/nav2z_client/custom_planners/pure_spinning_local_planner/CHANGELOG.rst
3838 ./smacc2_client_library/nav2z_client/custom_planners/backward_global_planner/CHANGELOG.rst
3826 ./smacc2_client_library/nav2z_client/custom_planners/forward_global_planner/CHANGELOG.rst
3828 ./smacc2_client_library/nav2z_client/custom_planners/forward_local_planner/CHANGELOG.rst
22138 ./smacc2_client_library/move_group_interface_client/CHANGELOG.rst
3885 ./smacc2_client_library/multirole_sensor_client/CHANGELOG.rst
3883 ./smacc2_client_library/ros_publisher_client/CHANGELOG.rst
3885 ./smacc2_client_library/ros_timer_client/CHANGELOG.rst
```

brettpac, 2025-06-10:
Thanks @tfoote, we're on it.

tfoote, 2025-06-11:
I've fixed it for the rosindex so it truncates content now https://github.com/ros-infrastructure/rosindex/pull/544

> only a few packages are really distributed in ros_index.
Only a few packages are distributed as binary packages, however the whole repository is included in the documentation: https://index.ros.org/r/SMACC2/github-robosoft-ai-SMACC2/#rolling 

For example https://index.ros.org/p/keyboard_client/ the CHANGELOG is now truncated. 

However of that changelog there's about 30 items of useful information int he CHANGELOG, and then a raw dump of what appears to be a github commit history. And alot of those are not useful descriptions for example "minor" or "format improvements" and there's some 3000+ lines of these non-useful commit log messages dumped into the CHANGELOG. Occasionally there's a reference to an issue in there, but that doesn't provide any more useful content either eg `  * update readme (#164)
` linking to https://github.com/robosoft-ai/SMACC2/pull/164 which only has content "update readme" 

> We could reduce this even more, and appart from this we will have a look to the Continous delivery scripts.

I don't think that this is an issue with regular updates causing things to grow. There's one PR adding 700k LOC and most of that is all pre history aka 0.0.0 content.  https://github.com/robosoft-ai/SMACC2/pull/364 

This is not a blocker anymore, but more of a piece of feedback that the CHANGELOGS for this project appear to be not useful for the purpose of understanding what's changed between versions for a user who wants to understand what's changed between versions. My broad recommendation would be to delete everything in the 0.0.0 categories and make sure that you're writing useful changelogs not just aggregating non-descriptive commit logs. And it appears that a large fraction of the history of the project is embedded into the CHANGELOGS of each of the example packages instead of just being the history of that package. For example https://github.com/robosoft-ai/SMACC2/pull/198 is referenced 6 times in the keyboard_client yet that commit appears to [not touch keyboard_client](https://github.com/robosoft-ai/SMACC2/pull/198/files) and potentially predate the existance of the package .

closed 2025-11-02 by brettpac